AndrejOrsula/pymoveit2 #4: UR robot support

tlpss, 2022-01-05:
Hi @AndrejOrsula, I was looking for a way to send target poses to a UR3e from python to Moveit2 and came across your work. 
I managed to get your bridge working with the ROS2 drivers for the UR by creating a new robot class: 

```
from typing import List

MOVE_GROUP_ARM: str = "ur_manipulator"
MOVE_GROUP_GRIPPER: str = "NA"

def joint_names(prefix: str = "") -> List[str]:
    return [
        prefix + "elbow_join",
        prefix + "shoulder_lift_joint",
        prefix + "shoulder_pan_joint",
        prefix + "wrist_1_joint",
        prefix + "wrist_2_joint",
        prefix + "wrist_3_joint",
    ]

def base_link_name(prefix: str = "") -> str:
    return prefix + "base_link"

def end_effector_name(prefix: str = "") -> str:
    return prefix + "tool0"

def gripper_joint_names(prefix: str = "") -> List[str]:
    return [
        prefix + "NA",
        prefix + "NA",
    ]

```
However the UR drivers do not include a gripper in their default URDF/SRDF, so I was wondering how you would want the UR class to handle this (so that i can make a PR).

AndrejOrsula, 2022-01-05:
Hello @tlpss and thanks for opening this issue.

To answer your question first, I'd just return `None`.
```diff
- from typing import List
+ from typing import List, Optional
```
```diff
- MOVE_GROUP_GRIPPER: str = "NA"
+ MOVE_GROUP_GRIPPER: Optional[str] = None
```
```diff
- def gripper_joint_names(prefix: str = "") -> List[str]:
-     return [
-         prefix + "NA",
-         prefix + "NA",
-     ]
+ def gripper_joint_names(prefix: str = "") -> Optional[List[str]]:
+    return None
```

---

However, I actually created the [panda config](https://github.com/AndrejOrsula/pymoveit2/blob/master/pymoveit2/robots/panda.py) purely as an example, such that all other examples can be used directly with [panda_ign_moveit2](https://github.com/AndrejOrsula/panda_ign_moveit2). Adding more and more robots would become unfeasible, where some robots might have multiple description packages that are incompatible with each other, e.g. `franka_description` from official `franka_ros` repo and `panda_description` from `moveit_resources` (even though it's exactly the same robot). By no means do I believe this is the optimal approach, so I usually just keep such config temporarily near the application/robot where I need it.

The proper approach would be to extract these information from semantic robot description (SRDF) passed into the node as parameter (URDF is also required for many robots, e.g. those that specify groups via chains). However, I have not attempted to do that yet as there are several complexities. I am not sure I ever will.

That said, I do not have any issue with you opening such a PR. Just a note this repo is meant to be temporary until official bindings are developed (hence I am not investing much time into it). I will be modifying it to my needs (semantic versioning is followed though).

tlpss, 2022-01-05:
Thanks for your reply. I'm aware that this is a temporary fix until Moveit2  has created the new bindings with pybind11. 

Myself, I was considering to create an artificial node in both python and c++ to still use the high-level move group interface but have not managed to get it to work so far, which is why I turned to your repo for a quick fix. 

>  (hence I am not investing much time into it). I will be modifying it to my needs (semantic versioning is followed though).

In  that case, if you don't mind, I might just fork the repo so that I do not have to worry about non-backward compatible changes.

AndrejOrsula, 2022-01-05:
Of course, fork it and modify as needed. :smile: 

Otherwise, just `clone`/`checkout` a compatible branch. Everything is `1.x.x` so far, so `1.1.0` is exactly as `1.0.0` with some extra backwards-compatible stuff. 

Once #3 is merged, `2.0.0` will be released. As an example, these changes are breaking because `PoseStamped` will be directly returned from `compute_fk()` instead of `GetPositionFK.Response`.

closed 2022-01-05 by tlpss

JaisonJose241, 2023-10-27:
Did anyone tried the moveit_py package present in the `main` branch of `moveit2` on ROS humble version?