AndrejOrsula/pymoveit2 #6: Cartesian Path for reaching a Pose Goal

slabua, 2022-01-17:
Currently, sending a new pose will starts the path planning but it is not in the form of a cartesian shortest path between the initial pose and the goal pose.
Is there a way to obtain such behaviour?

AndrejOrsula, 2022-01-17:
Hello @slabua,
Only kinematic paths (paths in joint space) and MoveIt Servo are currently supported within this repo. Interface for Cartesian paths is currently not implemented.

If you want to implement it yourself, a similar logic to kinematic path request could be added for the specific service for planning Cartesian paths with its corresponding type.

For certain applications, you could also use MoveIt Servo to achieve the desired motion. But that of course depends on what you need.

slabua, 2022-01-17:
Hi @AndrejOrsula, thank you for your reply.
Perhaps I haven't found yet where in ROS2 I could do that. I had previously used moveit_commander in ROS1 (and the UR robots).
I will investigate a little more, thanks again!

AndrejOrsula, 2022-01-17:
That's correct, `moveit_commander` is the Python interface for MoveIt (ROS 1). At the moment, there is no Python interface for MoveIt 2 (ROS 2).

This is the reason behind creation of this `pymoveit2` repo. It is temporary until the official Python bindings are implemented.

AndrejOrsula, 2022-04-08:
This feature is now added via https://github.com/AndrejOrsula/pymoveit2/pull/8 by @Schnilz. It is currently part of the `devel` branch, but it will become available on `master` from `2.1.0` release.

Thank you for your contribution.

PR #8 (2022-04-08): Add support for Cartesian paths
Hi,
thank you for writing this, especially with the good examples :)
I missed the cartesian planner, so i added it here. 
I don't know why the moveit2 interface is so weird, but the code works (and calls the same service as the GUI in rviz does i think).

closes #6
closed 2022-04-08 by AndrejOrsula