AndrejOrsula/pymoveit2 #17: Controlling real robot

metanav, 2022-08-08:
How can we use a real robot running microRos node (joint_states publisher and follow_trajectory_action server) and connected over serial to microRos Agent with your library?  Any example or hints would be helpful.

AndrejOrsula, 2022-08-08:
Hello,

There is no difference between simulated and real robots for `pymoveit2` interface. If you can use your MoveIt config with other interfaces (e.g. RViz 2 `MotionPlanning` plugin) to control your simulated/real robot, then you can also use this interface without reconfiguring your MoveIt config.

Please, take a look at [examples](https://github.com/AndrejOrsula/pymoveit2/tree/master/examples), e.g. [ex_pose_goal.py](https://github.com/AndrejOrsula/pymoveit2/blob/master/examples/ex_pose_goal.py), and see how the `MoveIt2` interface class is created. Instead of using `panda` defined in [panda.py](https://github.com/AndrejOrsula/pymoveit2/blob/master/pymoveit2/robots/panda.py), you can then use your own robot and define its groups, joints and links.

metanav, 2022-08-09:
Hi Andrej, thank you for your reply. I am a bit confused. The Ros2/MoveIt2 documentation is a bit vague and scattered into many different versions. Do I need to write/configure a launch script for ros controller to control a real robot? I could not find any useful documentation which have a minimal working setup or guidelines.

AndrejOrsula, 2022-08-09:
Hello,

Python module `pymoveit2` does not launch controllers, nor does it perform any configuration. You need to launch controllers and create the configuration yourself. If there is no implementation of [ros2_control](https://control.ros.org/master/index.html) for your robot, you might want to look into also implementing it yourself. For MoveIt 2, try following [tutorials](https://moveit.picknik.ai/humble/index.html). You can also try [MoveIt Setup Assistant 2.0](https://github.com/ros-planning/moveit2/tree/main/moveit_setup_assistant), which could simplify your setup. Try also looking at ROS Discourse or MoveIt Discord to see if there are better pointers than what I can provide with my limited knowledge.

I am closing this issue as it is unrelated to this repository. Python module `pymoveit2` does not differentiate between simulated and real robots. Both work equally. You must configure your robot (simulated or real) to function with MoveIt 2. Such configuration is outside the scope of this repo.

closed 2022-08-09 by AndrejOrsula