AndrejOrsula/pymoveit2 #18: get ee pose?

raysworld, 2022-08-27:
Hi, Andrej Orsula,

Many thanks for offering this convenient tool :-)

Could you kindly let me know if there is an easy way to get the ee pose of the robot, so that I can do something like:

```python
curr_pose = moveit2.get_ee_pose()
curr_pose.position.x = curr_pose.position.x + 0.1

moveit2.move_to_pose(position=curr_pose.position, quat_xyzw=curr_pose.quat_xyzw, cartesian=cartesian)
moveit2.wait_until_executed()
```

Any help is appreciated.

AndrejOrsula, 2022-08-27:
Hello @raysworld,

I believe there are multiple ways of achieving this.

1. [**`MoveIt2.compute_fk()`**](https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2.py#L667-L717) 
  - Its parameter `joint_state` defaults to current joint state; and parameter `fk_link_names` defaults to end-effector link name. Therefore, it returns the current pose of the end-effector wrt. base link, i.e. similar to your `get_ee_pose()`.  It can return `None` upon failure (if the service is not available/ready), so you should maybe put a check in place within your application.

```py
# Note: Returns `geometry_msgs.msg.PoseStamped`
curr_pose_stamped = moveit2.compute_fk()
curr_pose = curr_pose_stamped.pose
curr_pose.position.x += 0.1

moveit2.move_to_pose(position=curr_pose.position, quat_xyzw=curr_pose.quat_xyzw, cartesian=cartesian)
moveit2.wait_until_executed()
```

---

2. **You could use `tf2_ros.TransformListener` to get transformation from base frame to end-effector frame.**
  - Here is an example wrapper class that I use for this purpose [drl_grasping/envs/utils/tf2_listener.py](https://github.com/AndrejOrsula/drl_grasping/blob/c863a29098a63b7ae829fa0f150879e56fb7b8ac/drl_grasping/envs/utils/tf2_listener.py).

With this specific class, it could look something like this. But you can use `tf2_ros.TransformListener` and `tf2_ros.Buffer` directly if you want. Again, it can return `None` upon failure, so you should maybe put a check in place within your application.

```py
tf2_listener = Tf2ListenerStandalone(use_sim_time=False) # OR `Tf2Listener(node=existing_node)`
# Note: Returns `geometry_msgs.msg.Transform`
ee_transform = tf2_listener.lookup_transform_sync(
    source_frame=ee_link_name,
    target_frame=base_link_name,
)
ee_transform.translation.x += 0.1

moveit2.move_to_pose(position=ee_transform.translation, quat_xyzw=ee_transform.rotation, cartesian=cartesian)
moveit2.wait_until_executed()
```

---

3. **Specify `frame_id` in which planning should take place.**
  - Assumes that your end-effector is aligned with the coordinate in which you want it to move.
  - I did not test it, so I am not sure if it works out of the box.

```py
moveit2.move_to_pose(position=(0.1, 0.0, 0.0), frame_id=ee_link_name, cartesian=cartesian)
moveit2.wait_until_executed()
```

raysworld, 2022-08-29:
> 

@AndrejOrsula Great thanks to your detailed replies! gonna test every solution you mentioned🙏🙏🙏

raysworld, 2022-08-29:
Thanks again for the timely help.

Solution 1 & 2 works for me, besides minor revisions to Solution 1:
```python
# Note: Returns `geometry_msgs.msg.PoseStamped`
curr_pose_stamped = moveit2.compute_fk()
curr_pose = curr_pose_stamped[0].pose
curr_pose.position.x += 0.1

moveit2.move_to_pose(position=curr_pose.position, quat_xyzw=curr_pose.orientation, cartesian=cartesian)
moveit2.wait_until_executed()
```

As for solution 3, I tried it with the [panda robot ](https://github.com/AndrejOrsula/panda_ign_moveit2) but had no luck making it work and got the following error:
```log
[INFO] [1661773958.476435488] [ex_pose_goal]: Moving to {position: [0.1, 0.0, 0.0], quat_xyzw: [1.0, 0.0, 0.0, 0.0]}
[WARN] [1661773958.997006480] [ex_pose_goal]: Planning failed! Error code: -1.
[WARN] [1661773958.997979821] [ex_pose_goal]: Cannot execute motion because the provided/planned trajectory is invalid.
[WARN] [1661773958.998532384] [ex_pose_goal]: Cannot wait until motion is executed (no motion is in progress).
```
there might be some coordinate alignment issues🤔