AndrejOrsula/pymoveit2 #19: Order of joint names

metanav, 2022-09-02:
I have set the joints names in the robots/myrobot.py as follows:

```
def joint_names() -> List[str]:
    return [
        'base_joint',
        'elbow_joint',
        'shoulder_joint',
        'wrist_pitch_joint',
        'wrist_roll_joint',
        'gripper_joint'
    ]
```

But the follow_joint_trajectory message received by the controller has different order:

```
name:
- base_joint
- shoulder_joint
- elbow_joint
- wrist_pitch_joint
- wrist_roll_joint
- gripper_joint
```
So the example code: examples/ex_joint_goal.py follows which ones?

```
    # Declare parameter for joint positions
    node.declare_parameter(
        "joint_positions",
        [
            0.0,
            0.0,
            0.0,
            -0.7853981633974483,
            0.0,
            1.5707963267948966,
            0.7853981633974483,
        ],
    )
```

metanav, 2022-09-02:
Hi @AndrejOrsula,

The MoveIt2 initialization should use the same order which I specified in the robots/myrobot.py but the action server. gets different one.

```
   moveit2 = MoveIt2(
        node=node,
        joint_names=myrobot.joint_names(),
...
```

AndrejOrsula, 2022-09-02:
If you don't pass any `joint_names` to `MoveIt2.set_joint_goal()`, then the default values from constructor are used. This repository does not perform any sorting.

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2.py#L213

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2.py#L624-L626

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2.py#L632-L633

Does the index of the joint command match the corresponding joint name? If so, there should not be any issue.

metanav, 2022-09-02:
I can fix it by changing controller code which relies on order of the joint names list in the message header. Just wondering    MoveIt2 with RViz has alphabetical order and this library has the same order as in SRDF. The problem is: every time I switch to Rviz to this library I would need to change the controller code to parse the correct order.

AndrejOrsula, 2022-09-02:
You can try going through the code in this repo and printing debug statements with the joint names for service/action requests. I don't see a reason why they would be switched as a cause of this repository. If you can point to the specific line/code block/function causing it, I will take a look.

closed 2022-10-31 by AndrejOrsula