AndrejOrsula/pymoveit2 #20: Adding frame_id in header is absent

vuva, 2022-10-17:
I found some parts where the frame_id was not added to the header. Is it intentional for some reason? It could give wrong trajectory planning if you are planning with robots in a different frame than the default.
See my diff below to locate the missing parts.

Cheers,

```
diff --git a/pymoveit2/moveit2.py b/pymoveit2/moveit2.py
index c4a2db4..3987ff8 100644
--- a/pymoveit2/moveit2.py
+++ b/pymoveit2/moveit2.py
@@ -697,6 +697,7 @@ class MoveIt2:

         stamp = self._node.get_clock().now().to_msg()
         self.__compute_fk_req.header.stamp = stamp
+        self.__compute_fk_req.header.frame_id = "base_link"

         if not self.__compute_fk_client.wait_for_service(
             timeout_sec=wait_for_server_timeout_sec
@@ -994,6 +995,7 @@ class MoveIt2:

         stamp = self._node.get_clock().now().to_msg()
         self.__cartesian_path_request.header.stamp = stamp
+        self.__cartesian_path_request.header.frame_id = "base_link"

         self.__cartesian_path_request.path_constraints = (
             self.__move_action_goal.request.path_constraints
diff --git a/pymoveit2/moveit2_servo.py b/pymoveit2/moveit2_servo.py
index 5291c00..31a3f88 100644
--- a/pymoveit2/moveit2_servo.py
+++ b/pymoveit2/moveit2_servo.py
@@ -130,6 +130,7 @@ class MoveIt2Servo:

         twist_msg = deepcopy(self.__twist_msg)
         twist_msg.header.stamp = self._node.get_clock().now().to_msg()
+        twist_msg.header.frame_id = "base_link"
         twist_msg.twist.linear.x *= linear[0]
         twist_msg.twist.linear.y *= linear[1]
         twist_msg.twist.linear.z *= linear[2]
```

AndrejOrsula, 2022-10-31:
Hello @vuva and thank you for opening this issue.

The header for these is meant to be set only once during the initialisation of the class in order to improve performance as these operations are costly in Python. Some functions support changing them dynamically though.

For example, the header for the twist message under the MoveIt2Servo class is set here.

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2_servo.py#L73

Which is then later copied over to the particular published message in the following line (just above your diff addition).

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2_servo.py#L131

---

If you encounter a specific case where this is not the case (with a debug print of the published messages/requests), please let me know and I can take a look into it.

vuva, 2022-11-03:
Header in twist_msg is indeed not necessary but for __compute_fk_req and __cartesian_path_request it is mandatory. Without it moveit will compute trajectory in "world" frame. This has been tested in out setup (ROS2 Humble + Ubuntu 22.04 )

I will provide log as soon as we run the setup again.

AndrejOrsula, 2022-11-03:
The `header.frame_id` of  `self.__twist_msg` is set in the following line.

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2_servo.py#L73

---

The `header.frame_id` of `self.__compute_fk_req` is set in the following line.

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2.py#L1215

---

I do not see the `header.frame_id` of `self.__cartesian_path_request` being set anywhere in https://github.com/AndrejOrsula/pymoveit2/pull/8, so it indeed **needs to be added**. Currently, the implementation of Cartesian paths in this repository follows a different pipeline from all the other methods. It could be changed to make it more uniform, but I believe official Python bindings for MoveIt 2 will be released soon enough - so I won't spent time to refactor this repository.

If you want, you can open a PR where `self.__cartesian_path_request.header.frame_id` defaults to `self.__base_link_name` with an optional argument that changes it to a specific user-defined frame somewhere in the function below. Otherwise, let me know if you would prefer me to do it. Thank you.

https://github.com/AndrejOrsula/pymoveit2/blob/67f455fad35af0b69c4f754f1311175ac00cb7b0/pymoveit2/moveit2.py#L979-L983

vuva, 2022-11-20:
> If you want, you can open a PR where self.__cartesian_path_request.header.frame_id defaults to self.__base_link_name with an optional argument that changes it to a specific user-defined frame somewhere in the function below. Otherwise, let me know if you would prefer me to do it. Thank you.

What I did was simply using self.__base_link_name, which takes the base_link_name argument from Movit2 initialization or frame_id the function move_to_pose (I don't remember correctly). I think it would be sufficient since we will have the official Python binding for Moveit2 soon.

Thank you for your effort.

closed 2022-11-22 by AndrejOrsula