AndrejOrsula/pymoveit2 #23: Question about this repo with UR robots

marioesteban11, 2022-12-23:
Hi everyone,

I'm trying to bridge some movegroup functions to python I think that is not supported yet. 

And after a bit of research I saw this repo, and I was wondering if I can use this functions to move my UR robot (ur3e). I see that you can perfectly move panda robots, but i dont know how to bridge it to UR.

Thanks in advance

AndrejOrsula, 2022-12-23:
Hello @marioesteban11,

Try to go over the related (duplicate) issue: https://github.com/AndrejOrsula/pymoveit2/issues/4

Let me know if something else needs to be clarified after reading it. Good luck!

AndrejOrsula, 2022-12-23:
If you are lucky, one of the forks might already contain the setup for your robot. Try taking a look 🙂