AndrejOrsula/pymoveit2 #24: Wrong configuration of collision object

zhoubin-me, 2023-01-16:
In your code:
```python
        pose = Pose()
        pose.position = position
        pose.orientation = quat_xyzw
        msg.pose = pose
```
However, from [ROS2](http://docs.ros.org/en/melodic/api/moveit_msgs/html/msg/CollisionObject.html), it should be
```
msg.mesh_poses.append(pose)
```

AndrejOrsula, 2023-01-16:
Hello. Thank you for opening this issue.

The `CollisionObject.pose` is a valid field in ROS 2 for (`foxy`, **`galactic`**, `humble` and `rolling`), see the [official implementation on GitHub](https://github.com/ros-planning/moveit_msgs/blob/master/msg/CollisionObject.msg).

```
# The object's pose relative to the header frame.
# The shapes and subframe poses are defined relative to this pose.
geometry_msgs/Pose pose
```

The [link you shared](http://docs.ros.org/en/melodic/api/moveit_msgs/html/msg/CollisionObject.html) is for `melodic` (ROS 1).

zhoubin-me, 2023-01-18:
@AndrejOrsula Thanks, however if I use that, the planner seems ignore my objects

AndrejOrsula, 2023-01-18:
May I ask what ROS 2 distribution you use for your MoveIt 2? This repository was originally developed & tested with `galactic`.

zhoubin-me, 2023-01-20:
> May I ask what ROS 2 distribution you use for your MoveIt 2? This repository was originally developed & tested with `galactic`.

Foxy

AndrejOrsula, 2023-01-20:
Yeah, so I have not tested this package with `foxy`, so it is expected that some functionalities just won't work as intended. Everything should work in `galactic` (no longer supported), and most things should be usable in `humble` (although not fully tested).

If you would like this repository to support a `foxy`-specific branch, feel free to open a PR and I can make it target a new branch. I won't be testing the `foxy` code besides a minimal review, so it would be quick to merge.

AndrejOrsula, 2023-01-20:
I am closing this issue as it is related to use of unsupported distribution.

If you (or anyone) wants to add support for `foxy`, feel free to open a new PR with the changes that make this repo compatible with `foxy`. I would then take care of the rest and merge the PR into a new `foxy` branch.

closed 2023-01-20 by AndrejOrsula