AndrejOrsula/pymoveit2 #28: Joint States issues

Gaurang-1402, 2023-07-01:
Hi, I followed the instructions but the robot is not executing the planned motion as expected

Here is the error I get in the move_group.launch.py terminal

```
[ERROR] [1688223063.534889158] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached
[move_group-2] [WARN] [1688223160.048471449] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Unable to transform object from frame 'tool0' to planning frame'world' (Could not find a connection between 'world' and 'tool0' because they are not part of the same tree.Tf has two or more unconnected trees.)
[move_group-2] [WARN] [1688223160.048533739] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Unable to transform object from frame 'flange' to planning frame'world' (Could not find a connection between 'world' and 'flange' because they are not part of the same tree.Tf has two or more unconnected trees.)
[move_group-2] [WARN] [1688223160.048550778] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Unable to transform object from frame 'wrist_3_link' to planning frame'world' (Could not find a connection between 'world' and 'wrist_3_link' because they are not part of the same tree.Tf has two or more unconnected trees.)
[move_group-2] [WARN] [1688223160.048565094] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Unable to transform object from frame 'ft_frame' to planning frame'world' (Could not find a connection between 'world' and 'ft_frame' because they are not part of the same tree.Tf has two or more unconnected trees.)
[move_group-2] [ERROR] [1688223160.048647317] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-2] [ERROR] [1688223160.048668730] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-2] [ERROR] [1688223160.048743758] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-2] [ERROR] [1688223160.048881414] [moveit_robot_state.conversions]: Found empty JointState message
```

Here is the error in the ex_pose_goal terminal

```
$ ros2 run pymoveit2 ex_pose_goal.py --ros-args -p position:="[0.25, 0.0, 1.0]" -p quat_xyzw:="[0.0, 0.0, 0.0, 1.0]" -p cartesian:=False
[INFO] [1688223160.047524623] [ex_pose_goal]: Moving to {position: [0.25, 0.0, 1.0], quat_xyzw: [0.0, 0.0, 0.0, 1.0]}
[WARN] [1688223161.135484471] [ex_pose_goal]: Action server 'joint_trajectory_controller/follow_joint_trajectory' is not yet available. Better luck next time!
[WARN] [1688223161.136446016] [ex_pose_goal]: Cannot wait until motion is executed (no motion is in progress).

```

You can see the planned motion below but no execution

[image]

It tells me that /joint_states is not publishing messages

AndrejOrsula, 2023-07-13:
Hello,

> ```
> [WARN] [1688223161.135484471] [ex_pose_goal]: Action server 'joint_trajectory_controller/follow_joint_trajectory' is not yet available.
> ```

This would indicate that the controller is not running, so the issue is most likely related to the robot setup (`_description` and `_moveit_config`) rather than `pymoveit2`.

**What ROS 2 distribution do you use?** As [`README.md` of panda_ign_moveit2](https://github.com/AndrejOrsula/panda_ign_moveit2) indicates, it was tested with `galactic`. There is now [`humble` branch](https://github.com/AndrejOrsula/panda_ign_moveit2/tree/humble), but it is not fully tested.

If you want, you can also test the setup with prebuilt Docker images first to know how it should look like. For example:
- https://hub.docker.com/repository/docker/andrejorsula/ign_moveit2_examples
- https://hub.docker.com/repository/docker/andrejorsula/panda_ign_moveit2

JaisonJose241, 2023-10-27:
It's also observed that after some amount of time or with good computation PCs, it's possible to continue with same request. And weird thing is in the same instance it runs without error.

brennand, 2024-07-09:
Is this because of this bug?  

`for joint_name in self.joint_names:`

Where `self.joint_names`  should be `self.__joint_names`

AndrejOrsula, 2025-08-16:
We are closing this issue due to inactivity. Please feel free to comment and reopen it if anyone has a similar problem.

closed 2025-08-16 by AndrejOrsula