AndrejOrsula/pymoveit2 #29: move_to_configuration's cartesian flag seems unnecessary

amalnanavati, 2023-07-12:
#8 added a `cartesian` flag to `move_to_configuration(...)`. However, setting that flag to `True` (e.g., by modifying [the `move_to_configuration` call in `ex_joint_goal.py`](https://github.com/AndrejOrsula/pymoveit2/blob/29dc040ed5fd8703bb19795474114ada2a79d6b0/examples/ex_joint_goal.py#L63)) fails with an IndexError. The reason it fails it that the cartesian planner expects position and orientation goal constraints ([here](https://github.com/AndrejOrsula/pymoveit2/blob/29dc040ed5fd8703bb19795474114ada2a79d6b0/pymoveit2/moveit2.py#L1015) and [here](https://github.com/AndrejOrsula/pymoveit2/blob/29dc040ed5fd8703bb19795474114ada2a79d6b0/pymoveit2/moveit2.py#L1021)), whereas `move_to_configuration(...)` only sets joint goal constraints ([here](https://github.com/AndrejOrsula/pymoveit2/blob/29dc040ed5fd8703bb19795474114ada2a79d6b0/pymoveit2/moveit2.py#L329C21-L329C21)).

The only way to rectify this is to set position and orientation goal constraints _before_ calling `move_to_configuration(...)`. However, at that point, it seems better to either use `move_to_pose(...)` (if the user only cares about pose goal constraints) or make a custom call to `plan(...)` (if the user wants to combine pose and joint constraints).

Although MoveIt2's RVIZ interface allows for cartesian planning with a joint configuration as the goal, as far as I can tell it first uses FK to convert the goal configuration into a goal end effector pose, and then computes a cartesian plan with that end effector pose as the goal constraint.

I see three options for addressing this:
1. Remove the `cartesian` flag from `move_to_configuration(...)`.
2. If `move_to_configuration(...)` is called with the `cartesian` flag, first compute FK and then set position/orientation constraints based on the end effector pose (as opposed to setting joint constraints).
3. Keep the code as-is, but add to the docstring of `move_to_configuration(...)` that the `cartesian` flag only works if position and orientation goal constraints have been set beforehand.

I'm happy to make the changes, but before diving in I wanted to get others' thoughts on this issue and the best way to address it.

AndrejOrsula, 2023-07-13:
Hello @amalnanavati,

Nice catch!

I believe all three options are valid. The second would be preferred, in my opinion.

1. It seems like a rare use case to plan a Cartesian path specified by goal configuration in the joint space. So removing the `cartesian` flag from `move_to_configuration()`  probably won't affect many people (if anyone at all). However, this would introduce a breaking change regardless - but we can make a new major release.
2. Yes, computing FK first seems like the proper way of doing it, and the result would be aligned with the expected behaviour.
3. This is a good alternative to the second option without introducing breaking changes.

It would be great if you are interested in contributing! I would be happy to merge the changes and then make a new release.
But FYI: you might be better off using the [official Python bindings](https://picknik.ai/moveit/ros/python/google/2023/04/28/GSOC-MoveIt-2-Python-Bindings.html) mentioned in the README. I did not try it myself or looked at the interface yet - but I assume it is done more properly than this repository.

amalnanavati, 2023-07-15:
Thanks for weighing in, and good point re. the official Python bindings. I’ll see how they do this before moving forward.