AndrejOrsula/pymoveit2 #31: Use different planner id (PILZ)

mataruzz, 2023-07-21:
Hi, 
I'm using your library to move a robotic arm. For some reason using the default planner I have problems in the planning. I would use the "_pilz_industrilal_motion_planner_" as pipeline_id but I found it only inside "_moveit2.py_" (line 1182) and it is commented.

What would you suggest to change the default planner and set pilz? Is it possible without changing the source?

AndrejOrsula, 2023-07-21:
Hello,

Have you tried configuring the default planner in the `*_moveit_config` package of your robot? The place and how you change it would depend on how your config/launch scripts are structured. Try consulting the MoveIt 2 documentation, as that would be up-to-date.

I have not tried it myself, but I believe you would generally need to change parameters `planning_pipelines` and `default_planning_pipeline` such that they point to the pilz configuration (assuming you have it configured and the plugin is installed on your system).

amalnanavati, 2024-03-26:
@mataruzz I believe #48 resolves this.