AndrejOrsula/pymoveit2 #44: Force control

albic98, 2024-01-11:
How would I add Force_checker to pymoveit2.py script so that when I have an 3D uneven object on which I want to go from one point to another with -10N force I could do it with just using ex_pose_goal.py?

JaisonJose241, 2024-01-11:
You can sense the force using topic. Why do you require pymoveit2 anyways?

albic98, 2024-01-11:
I am using it to move from one cartesian pose to another, but I have uneven 3D surface and I have to move from one point to the other with 10N force on the surface. Is there a better solution than pymoveit2? I am using franka_robot_state_broadcaster for sensing.

albic98, 2024-01-11:
I tried writing a script which uses o_f_ext_hat_k for sensing but still the robot moves from one point to the other with no force adjustments. This is my first time using ROS2 and Franka Emika. Can you help?

AndrejOrsula, 2024-01-13:
Hello @albic98, I am afraid `pymoveit2` does not provide any functionality that is required for reliable force-based control. The underlying MoveIt2 is a fantastic planning framework, but it mostly focuses on kinematics. You could potentially utilize MoveIt Servo to come up with some hacky solution while monitoring the pose/force, but that is very far from optimal.

I would suggest you search elsewhere. Going over the documentation of the hardware platform you use (Franka Emika) is where I would personally begin. Good luck!

closed 2024-01-13 by AndrejOrsula