AndrejOrsula/pymoveit2 #45: Cartesian speed

albic98, 2024-01-15:
How do can I change Cartesian speed for Franka Emika with this script? I tried changing "max_cartesian_speed" parameter, but it does not change anything. Can you help?

AndrejOrsula, 2024-01-15:
Hello @albic98,

That sounds like a bug. I think this might be because the functionality is simply not implemented for `execute_via_moveit=False`. You could try with `execute_via_moveit=True` and see if it changes anything. 

Alternatively, you could try taking a look at [moveit2.py](https://github.com/AndrejOrsula/pymoveit2/blob/6268b9993352b408e268b18856792deecb1bcffa/pymoveit2/moveit2.py) and see how [`@max_cartesian_speed.setter`](https://github.com/AndrejOrsula/pymoveit2/blob/6268b9993352b408e268b18856792deecb1bcffa/pymoveit2/moveit2.py#L1693-L1695) needs to be modified to fix this. PRs are welcome. :slightly_smiling_face:

albic98, 2024-01-16:
I tried switching execute_via_moveit to True, but it keeps moving in joint path, not cartesian. Also I tried everything with the setter, but still no solution.

AndrejOrsula, 2024-04-25:
Addressed by https://github.com/AndrejOrsula/pymoveit2/pull/53

closed 2024-04-25 by AndrejOrsula