AndrejOrsula/pymoveit2 #55: MoveGroup Goal request missing attribute

alek5k, 2024-03-28:
Hi there,

Thanks for the library - found it so much easier than trying to get the MoveIt API working. I came across a bug (?) which seems to happen during initialization. The function `__init_move_action_goal` creates a request with a cartesian speed parameter. This parameter is missing in my request. Syntax highlighting seems to not find it either (see picture below)

[image]

I am using ROS Iron.

I've commented out that line and the code works fine (however I have not tested robot movements - just using this library to attach collision objects).

AndrejOrsula, 2024-03-29:
Hello @alek5k. Yes, this change in Iron was noted via https://github.com/AndrejOrsula/pymoveit2/pull/49.

https://github.com/AndrejOrsula/pymoveit2/pull/56 should address it once it gets merged in.

alek5k, 2024-04-03:
Hi @AndrejOrsula 

Ah great, I missed that one. Should I use the `devel` branch for iron? I noticed there was an `iron_compatibility` branch but seems to have been deleted now.

Thanks again for the library, do you plan on maintaining it long term? I found it much easier than using the official MoveIt API - no need to fiddle with robot definitions, which was proving extremely difficult with a UR10.

AndrejOrsula, 2024-04-03:
Hello @alek5k,

> Should I use the `devel` branch for iron?

Yes, please try using the `devel` branch. I have not tested it on `iron`, so it would be nice to get someone's confirmation before it gets merged into the default branch.

> I noticed there was an `iron_compatibility` branch but seems to have been deleted now.

The commits of the `iron_compatibility` branch were merged into the `devel` branch via https://github.com/AndrejOrsula/pymoveit2/pull/56, so there is no need for it anymore.

> do you plan on maintaining it long term?

For now, I will continue to review & merge PRs and answer issues. However, I do not actively use `pymoveit2` anymore, so I won't actively develop new features or make any extensive modifications myself. It is used in some projects, so maintaining backwards-compatibility in a single branch for multiple distributions would be ideal.

I still haven't tried the official MoveIt 2 Python bindings because I work on other things, but it is what I generally recommend people try first if they can.

alek5k, 2024-04-05:
Thanks @AndrejOrsula I will use the `devel` branch!

I think your library here fills the purpose that `moveit_commander` did in ROS1. I wonder if it's possible to ask the ROS guys to merge it into ROS officially? Maybe that way others can actively develop it if they prefer to use the action interface instead of the API? I'm not sure what the pros/cons are though.

Btw I am starting a PhD focusing on DRL-based robotic grasping and I noticed your other repo looking at this problem. Perhaps our paths may intersect again in future :)

closed 2024-04-05 by alek5k

AndrejOrsula, 2024-04-08:
> I wonder if it's possible to ask the ROS guys to merge it into ROS officially? Maybe that way others can actively develop it if they prefer to use the action interface instead of the API?

I am not sure. I have not discussed this with any of the MoveIt maintainers, but it might cause additional confusion for end users if multiple "official" libraries serve a very similar purpose. If the motive is to have a simpler installation process, then supporting something like this shouldn't be too difficult.

```bash
pip install pymoveit2 @ git+https://github.com/AndrejOrsula/pymoveit2.git
```

> Btw I am starting a PhD focusing on DRL-based robotic grasping and I noticed your other repo looking at this problem. Perhaps our paths may intersect again in future :)

That's great! Yeah, it might happen. :slightly_smiling_face: Good luck with shaping your research topic!