AndrejOrsula/pymoveit2 #61: No IK solution found for round-off values of position and orientation

emp002, 2024-04-18:
Hello,
I am trying to control a 4dof scara robot with this package. Below I have attached the model of my robot. 
First, I tried using ex_fk.py to get end effector position by giving joint angles as arguments. 
Then, I used the same values with 12 decimal places to get joint angles using ex_ik.py. It works perfectly, it gives joint angles close to [0, 0, 0, 0] but with decimal places.
Then again, I used the same values but rounded-off to 1 decimal. I am getting an error that "IK computation failed! Error code: -31."

I have also attached the screenshot of my robot description terminal and the ex_ik launch terminal.
What could be the reason that it is not calculating IK with less decimal places?

[image]

[image]
[image]

AndrejOrsula, 2024-04-25:
Hello,

You could try increasing the tolerances of relevant methods (refer to `tolerance`, `tolerance_position` and `tolerance_orientation` arguments). But I am unsure if all planners (MoveIt) support them. Good luck!

AndrejOrsula, 2025-08-16:
We are closing this issue due to inactivity. Please feel free to comment and reopen it if anyone has a similar problem.

closed 2025-08-16 by AndrejOrsula