AndrejOrsula/pymoveit2 #72: get pre_configurated joint from moveit conifg

adamanov, 2024-06-03:
Hello, first of all, thanks for the cool repo, and for providing the Python binding with Moveit on Humble. 

in moveit1 there was the following two function
```
arm = MoveGroupCommander("arm")
arm.set_named_target("arm_home_name")
arm.go()

```
by setting the `arm_home_name`  referring  joints configuration inside the `.srdf ` file, it was super easy to move to a pre-defined target pose.

I was searching, a way how to do it with help of your package `pymoveit2`, and I found only the `move_to_configuration()` function which allows moving to a joint position, however, I am not able to move directly to get joint angles which I defined inside `.srdf`, is there any way to get them from moviet config? without hard- coding inside the my application node?

All best

AndrejOrsula, 2024-06-04:
Hello @adamanov,

The codebase does not actually parse URDF/SRDF configs and afaik MoveIt 2 does not expose the list of named configurations via ROS 2. Therefore, there is currently no way to use them through `pymoveit2`.

adamanov, 2024-06-07:
> Hello @adamanov,
> 
> The codebase does not actually parse URDF/SRDF configs and afaik MoveIt 2 does not expose the list of named configurations via ROS 2. Therefore, there is currently no way to use them through `pymoveit2`.

thanks for reply, i ended up also migrated on ros2 rolling to install moveit2 with python binding...

closed 2024-06-07 by adamanov