AndrejOrsula/pymoveit2 #73: wait_until_executed does not wait when planning is not finished yet

Kaiser1401, 2024-06-10:
`wait_until_executed` only waits when the plan has been send for execution, but when the planning takes longer, there is no execution yet and the function returns. I think that might be what happens in #28 as well.

Found while working on Jazzy-Harmonic adaptation

(
https://github.com/AndrejOrsula/panda_ign_moveit2/pull/34
https://github.com/AndrejOrsula/ign_moveit2_examples/pull/27
)

AndrejOrsula, 2024-06-10:
I wonder what use case could trigger this. The internal code waits since the request was called (`self.__is_motion_requested`) until the execution ends (`self.__is_executing`). As far as I remember, `self.__is_motion_requested` should be unset after execution begins and `self.__is_executing` is set.

https://github.com/AndrejOrsula/pymoveit2/blob/afd30fa01724ba890599ae6143586d6694ff2319/pymoveit2/moveit2.py#L745-L759

Do you know which call triggers this behaviour? That could help in pointing out the flaw in the logic somewhere.

Kaiser1401, 2024-06-11:
I think I was a bit confused and thought the execute(plan(..)) nesting somewhat would get messed up but can now not find why it would. I'll try to provocate the error again and check where things go sideways

AndrejOrsula, 2025-08-16:
We are closing this issue due to inactivity. Please feel free to comment and reopen it if anyone has a similar problem.

closed 2025-08-16 by AndrejOrsula