AndrejOrsula/pymoveit2 #78: Kinova Gen 3 Lite

htchr, 2024-10-23:
Hello, 

I'm attempting to modify the `ex_pose_goal.py` example to work with a Kinova Gen 3 Lite robot arm in ros2 Humble.
So far, the script appears to run fine, but the robot arm doesn't move.
Below are the modified script and terminal outputs:

(modified) ex_pose_goal.py
```python
#!/usr/bin/env python3
"""
Example of moving to a pose goal.
- ros2 run pymoveit2 ex_pose_goal.py --ros-args -p position:="[0.25, 0.0, 1.0]" -p quat_xyzw:="[0.0, 0.0, 0.0, 1.0]" -p cartesian:=False
- ros2 run pymoveit2 ex_pose_goal.py --ros-args -p position:="[0.25, 0.0, 1.0]" -p quat_xyzw:="[0.0, 0.0, 0.0, 1.0]" -p cartesian:=False -p synchronous:=False -p cancel_after_secs:=1.0
- ros2 run pymoveit2 ex_pose_goal.py --ros-args -p position:="[0.25, 0.0, 1.0]" -p quat_xyzw:="[0.0, 0.0, 0.0, 1.0]" -p cartesian:=False -p synchronous:=False -p cancel_after_secs:=0.0
"""

from threading import Thread

import rclpy
from rclpy.callback_groups import ReentrantCallbackGroup
from rclpy.node import Node

from pymoveit2 import MoveIt2, MoveIt2State

def main():
    rclpy.init()

    # Create node for this example
    node = Node("ex_pose_goal")

    # Declare parameters for position and orientation
    node.declare_parameter("position", [-0.018, -0.261, 0.629])
    node.declare_parameter("quat_xyzw", [0.238, 0.534, 0.373, 0.720])
    node.declare_parameter("synchronous", True)
    # If non-positive, don't cancel. Only used if synchronous is False
    node.declare_parameter("cancel_after_secs", 0.0)
    # Planner ID
    node.declare_parameter("planner_id", "RRTConnectkConfigDefault")
    # Declare parameters for cartesian planning
    node.declare_parameter("cartesian", True)
    node.declare_parameter("cartesian_max_step", 0.0025)
    node.declare_parameter("cartesian_fraction_threshold", 0.0)
    node.declare_parameter("cartesian_jump_threshold", 0.0)
    node.declare_parameter("cartesian_avoid_collisions", False)

    # Create callback group that allows execution of callbacks in parallel without restrictions
    callback_group = ReentrantCallbackGroup()

    moveit2 = MoveIt2(
        node=node,
        joint_names=['joint_1',
                     'joint_2',
                     'joint_3',
                     'joint_4',
                     'joint_5',
                     'joint_6',
                     'right_finger_bottom_joint'],
        base_link_name='base_link',
        end_effector_name='end_effector_link',
        group_name='gen3_lite_arm',
        callback_group=callback_group,
    )
    moveit2.planner_id = (
        node.get_parameter("planner_id").get_parameter_value().string_value
    )

    # Spin the node in background thread(s) and wait a bit for initialization
    executor = rclpy.executors.MultiThreadedExecutor(2)
    executor.add_node(node)
    executor_thread = Thread(target=executor.spin, daemon=True, args=())
[... 248 lines cut ...]
[rviz2-2]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-2] [ERROR] [1729722068.819627818] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-2] [INFO] [1729722068.830486193] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-2] [INFO] [1729722068.882718352] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[rviz2-2] [INFO] [1729722068.882745409] [moveit_robot_model.robot_model]: Loading robot model 'gen3_lite'...
[rviz2-2] [INFO] [1729722068.882751615] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-2] [WARN] [1729722068.990723441] [moveit_robot_model.robot_model]: Could not identify parent group for end-effector 'gen3_lite_gripper'
[rviz2-2] [INFO] [1729722069.347833383] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-2] [INFO] [1729722069.348285102] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-2] [INFO] [1729722069.494123105] [interactive_marker_display_99646268849504]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-2] [INFO] [1729722069.514200495] [interactive_marker_display_99646268849504]: Sending request for interactive markers
[rviz2-2] [INFO] [1729722069.547355008] [interactive_marker_display_99646268849504]: Service response received for initialization
[rviz2-2] [INFO] [1729722069.581988522] [moveit_ros_visualization.motion_planning_frame]: group gen3_lite_arm
[rviz2-2] [INFO] [1729722069.582002730] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'gen3_lite_arm' in namespace ''
[rviz2-2] [INFO] [1729722069.585610406] [move_group_interface]: Ready to take commands for planning group gen3_lite_arm.
[rviz2-2] [INFO] [1729722069.586144628] [moveit_ros_visualization.motion_planning_frame]: group gen3_lite_arm
```

Terminal output of pymovit2:
```
vscode@legion24:/kinova_ws$ ros2 run t516_lab7 ex_pose_goal 
[INFO] [1729722075.391154304] [ex_pose_goal]: Moving to {position: [-0.018, -0.261, 0.629], quat_xyzw: [0.238, 0.534, 0.373, 0.72]}
```

Thank you,

htchr, 2024-10-25:
I have found a solution:

I use this command to launch the Kinova arm in simulation:

```
ros2 launch kortex_bringup kortex_sim_control.launch.py sim_ignition:=false sim_gazebo:=true robot_type:=gen3_lite robot_name:=gen3_lite dof:=6 gripper:=gen3_lite_2f use_sim_time:=true launch_rviz:=false
```

I use this launch file instead to run MoveIt2 with the Kinova arm:

```
ros2 launch kinova_gen3_lite_moveit_config move_group.launch.py use_sim_time:=true
```

Even though both simulation files set `use_sim_time:=true`, I have to manually reconfigure move_group and moveit_simple_controller_manager to use simulation time in rqt.

closed 2024-10-25 by htchr

Averykla, 2024-11-23:
Are you able to change the velocity by changing the value of moveit2.max_velocity (when cartesian is set to true) in your humble version of ros2? I can't change the velocity in this way in my humble version of ros2. @htchr