AndrejOrsula/pymoveit2 #79: Question about cartesian value in ex_pose_goal.py

Averykla, 2024-11-22:
Hi author, I need to set a pre-grasp point during the robot grasping process, and then let the robot go down vertically with the posture of the pre-grasp point after the robot reaches the pre-grasp point to grasp the object. So I set the value of cartesian in your program to true during the process of the robot moving from the pre-grasp point to the grasping point. But after my test, I found that as long as I set the value of cartesian to true, my robot will move very fast. Is there any value in the program that can constrain the Cartesian path speed?

  I tried cartesian_max_step, but the effect was not obvious. In addition, I would like to ask you a question. When my robot reaches the pre-grasp point, if I continue to use the original RRT planning algorithm, the robot will execute a very complex trajectory, sometimes going around in a circle and then going down to grasp the object. What path planning should I choose from the pre-grasp point to the grasping point?

Averykla, 2024-11-22:
I read this <Added setting of cartesian speed and acceleration #53>.
I'm so Sorry, I don't have a deep understanding of MoveIt, so I still don't understand how to set the speed and acceleration values ​​of the Cartesian path.

Averykla, 2024-11-26:
Hi @AndrejOrsula , I found that modifying `move_action_goal.request.max_cartesian_speed = 0.0 `didn't work either. I think it's probably because the humble `class GetCartesianPath_Request `doesn't have speed and acceleration attributes.
So, I think this may not be an easy problem.

AndrejOrsula, 2024-11-27:
I checked, and you are correct. I took a quick look, but unfortunately, I didn't figure out the origin of the issue (internal/external). If you find it, please let us know. :slightly_smiling_face:

AndrejOrsula, 2026-09-16:
The latest release [**5.0.0**](https://github.com/AndrejOrsula/pymoveit2/tree/5.0.0) resolves this problem. Feel free to reopen this issue if necessary. Thank you for your patience.

closed 2026-09-16 by AndrejOrsula