AndrejOrsula/pymoveit2 #84: Cannot execute motion because the provided/planned trajectory is invalid

gemixin, 2025-02-25:
I have installed pymoveit2 in a new workspace (Ubuntu 22.04). I am using ROS2 Humble which is sourced automatically in my bashrc. 

I launch my UR5e sim using the following commands (in terminals with my moveit workspace sourced):
`ros2 launch ur_robot_driver ur5e.launch.py robot_ip:=localhost use_fake_hardware:=true launch_rviz:=false`
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true`

This loads my UR5e robot as expected. I can move the joints using a C++ node as per the MoveIt2 tutorials.

I wanted to try PyMoveIt2 so in a new terminal, I source my pymoveit2 workspace and run:
`ros2 run pymoveit2 ex_joint_goal.py --ros-args -p joint_positions:="[1.57, -1.57, 0.0, -1.57, 0.0, 1.57, 0.7854]"`

I get the following output:
`
[INFO] [1740489056.989646014] [ex_joint_goal]: Moving to {joint_positions: [1.57, -1.57, 0.0, -1.57, 0.0, 1.57, 0.7854]}
[WARN] [1740489057.000986855] [ex_joint_goal]: Service 'plan_kinematic_path' is not yet available. Better luck next time!
[WARN] [1740489057.008722221] [ex_joint_goal]: Cannot execute motion because the provided/planned trajectory is invalid.
[WARN] [1740489057.020831590] [ex_joint_goal]: Cannot wait until motion is executed (no motion is in progress).
`
I've tried with different positions and also tried the pose example, but I can't get past this issue.

I also tried with the Panda robot but the outcome was the same:
`ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_moveit_config_demo_empty.rviz` 

Any help greatly appreciated.

NoNoid, 2025-02-25:
Just came here to open the same issue, guess its not just me.

Vinay-Balaji, 2025-02-28:
Same here @AndrejOrsula 

```

(back_massage_bot) vinay@balaji:/ws/src/main_ros$ ros2 run pymoveit2 ex_pose_goal.py --ros-args -p position:="[-0.272, 0.003, 0.411]" -p quat_xyzw:="[0.847, 0.526, -0.070, -0.015]" -p cartesian:=True
[INFO] [1740784207.353443551] [ex_pose_goal]: Moving to {position: [-0.272, 0.003, 0.411], quat_xyzw: [0.847, 0.526, -0.07, -0.015]}
[WARN] [1740784207.456574000] [ex_pose_goal]: Planning failed! Cartesian planner completed 0.013468013468013467 of the trajectory, less than the threshold 0.99.
[WARN] [1740784207.457674857] [ex_pose_goal]: Cannot execute motion because the provided/planned trajectory is invalid.
[WARN] [1740784207.459096531] [ex_pose_goal]: Cannot wait until motion is executed (no motion is in progress).
(back_massage_bot) vinay@balaji:/ws/src/main_ros$ 

```

For a pose that works from the rviz/moveit gui

Vinay-Balaji, 2025-02-28:
This can be resolved by waiting for the service as referenced in

https://github.com/AndrejOrsula/pymoveit2/issues/86

AndrejOrsula, 2025-03-01:
Thank you all for addressing this issue!

Indeed, it appears to be related to the handling of the service. Unfortunately, I am unable to reproduce it with my current setup ~ so I would appreciate it if one of you could contribute a patch that fixes it for you. Thank you in advance.

AndrejOrsula, 2025-03-04:
Many thanks to @belalhmedan90, who offered a fix for this issue via https://github.com/AndrejOrsula/pymoveit2/pull/87!

The PR has been merged into the default branch. Please let us know if the issue persists. Thanks.

closed 2025-08-16 by AndrejOrsula