AndrejOrsula/pymoveit2 #86: Few issues in MoveIT interfacing

belalhmedan90, 2025-02-28:
Hi, thanks for the great effort, I noticed multiple issues in the implementation, and I want to point them out:
1. `rate.sleep()` on some machines sleeps forever, it might be better to change it to `rclpy.spin_once(self._node, timeout_sec=1.0)`
2. Waiting for service before checking availability: `self.__service_client.wait_for_service(timeout_sec=3.0)`.
3. Starting state is `None`, I have to weak the code as:
```py
        # Define starting state for the plan (default to the current state)
        while(start_joint_state is None):
            self._node._logger.warn(message="Joint states are not avaliable yet!")
            if(self.__joint_state is not None):
                start_joint_state = self.__joint_state
                break
            else:
                rclpy.spin_once(self._node, timeout_sec=1.0)
        self._node._logger.info(message="Joint states are avaliable now")
```
If you please consider these changes it might enhance the API imo, thanks.

AndrejOrsula, 2025-03-01:
Hi @belalhmedan90, thank you very much for opening this issue!

Would you be willing to contribute a PR that addresses these changes?
Thank you in advance for your consideration.

belalhmedan90, 2025-03-03:
> Hi [@belalhmedan90](https://github.com/belalhmedan90), thank you very much for opening this issue!
> 
> Would you be willing to contribute a PR that addresses these changes? Thank you in advance for your consideration.

Sure, please see PR #87

AndrejOrsula, 2025-03-04:
https://github.com/AndrejOrsula/pymoveit2/pull/87 is now merged

closed 2025-03-04 by AndrejOrsula