AndrejOrsula/pymoveit2 #92: upstreaming and bloom release

christian-rauch, 2025-04-28:
Thanks for this awesome package. By using the MoveIt2 actions and services provided by the move_group on the robot, this is the MoveIt "client-side" Python API in the spirit of the ROS 1 API that I was looking for.

Do you think it would be possible to "upstream" your Python API to https://github.com/moveit as a "Python client-side API" alternative to `moveit_py`?

Are you planning to bloom/release your package to the ROS repo?

AndrejOrsula, 2025-05-05:
Hello @christian-rauch,

Thank you for the kind words and for raising these points!

> Do you think it would be possible to "upstream" your Python API to https://github.com/moveit as a "Python client-side API" alternative to `moveit_py`?

This has been discussed multiple times (even before `moveit_py`), with the most recent mention [here](https://github.com/moveit/moveit_task_constructor/issues/647#issuecomment-2619945410). However, these discussions have never led anywhere. Frankly, by observing the maintenance activity across some projects within that org, I am not so sure I like the idea anymore...

> Are you planning to bloom/release your package to the ROS repo?

No, I have not actively considered bloom. Do you think this project is worth it? I could take a look at it sometime at the end of the month.

I had thought about adding a basic `pyproject.toml` to make it installable via Python package managers (e.g., via direct git url), and then maybe publish it to PyPI. However, I never checked whether it is really a good idea or not.

christian-rauch, 2025-05-05:
> > Are you planning to bloom/release your package to the ROS repo?
> 
> No, I have not actively considered bloom. Do you think this project is worth it? I could take a look at it sometime at the end of the month.

I definitely think this package is worth providing a bloomed release. I used `pymoveit2` before switching to `moveit_py` because I thought that the "official" bindings provide better support. But using a separate MoveIt instance for the Python bindings always caused issues in previous projects, and I am now again a happy user of `pymoveit2`.

I can see the benefit of running your own MoveIt instance in Python, if there is only a single instance controlling the robot, and you do not want the overhead of having to communicate with a MoveIt2 server. But once you have multiple processes wanting to use MoveIt2, making sure that all instances share the same configuration and planning scene is a huge overhead and very error-prone. `pymoveit2` resembles the "old" MoveIt behaviour in ROS 1 with one single MoveIt instance with centralised configuration and planning scene. IMHO, this should have been the approach to MoveIt2 Python bindings in the first place.

> I had thought about adding a basic `pyproject.toml` to make it installable via Python package managers (e.g., via direct git url), and then maybe publish it to PyPI. However, I never checked whether it is really a good idea or not.

Hosting Python packages on PyPI also involves hosting all its dependencies. That means you also have to package a large part of the ROS 2 Python API (rclpy, action_msgs, moveit_msgs, ...).

If you are interested in blooming `pymoveit2`, I could help with that. I would probably start with https://github.com/jspricke/ros-deb-builder-action to check if the package dependencies are properly specified, and it can be installed.

AndrejOrsula, 2025-05-11:
> Hosting Python packages on PyPI also involves hosting all its dependencies. That means you also have to package a large part of the ROS 2 Python API (rclpy, action_msgs, moveit_msgs, ...).

You are right, it is definitely not ideal.

> If you are interested in blooming `pymoveit2`, I could help with that. I would probably start with https://github.com/jspricke/ros-deb-builder-action to check if the package dependencies are properly specified, and it can be installed.

Thank you for the suggestion. I added the CI workflow in https://github.com/AndrejOrsula/pymoveit2/pull/97, and it seems to pass for `humble`, `jazzy`, and `noble` ~ but it does not really test anything besides installing this package. Currently, `pymoveit2` does not have any tests, so it might be difficult to get much more use out of that workflow.

I would appreciate your help with the bloom release. But I won't really have time for it until next month.

christian-rauch, 2025-05-11:
> I would appreciate your help with the bloom release. But I won't really have time for it until next month.

The first time procedure to release a new package is quite involved. You first need to become member of a (new) release team and then create a new release repo. Then, for each new tagged release, you create a release via `bloom-release`, which pushes to your new release repo. Once this has been reviewed and accepted, the build farm is finally building your package, which eventually gets released to the Debian repo. Ideally, you already have a working Debian package by then, e.g. via the CI action I suggested, to save you from iterating on the bloom-release, review and build farm cycle.

See the details here: https://docs.ros.org/en/jazzy/How-To-Guides/Releasing/First-Time-Release.html

Once you start with this and create a new "release team", you can ask to have me added to the team as well. This means I will have write access to the release repo and can create new releases via `bloom-release` too.

AndrejOrsula, 2025-06-08:
Thank you, @christian-rauch, for explaining the procedure and all your effort in setting up the [CI action](https://github.com/AndrejOrsula/pymoveit2/pull/97)!

I'm wondering if you have some suggestions for supporting the release across multiple ROS distributions. Would it just involve creating multiple branches under this repo (humble, jazzy, ...), and then adding the appropriate entries under https://github.com/ros/rosdistro?

AndrejOrsula, 2025-06-08:
Thank you for your suggestion!

Do you think it would be simpler to maintain this repo with "switches" rather than separate branches? Thinking about https://github.com/AndrejOrsula/pymoveit2/pull/101 as an example. If so, could you please point me to a repository with a Python project that uses a similar logic? Thank you in advance.

christian-rauch, 2025-08-11:
I started the process:
- index to rosdistro: https://github.com/ros/rosdistro/pull/47271
- release team: https://github.com/ros2-gbp/ros2-gbp-github-org/issues/846

christian-rauch, 2025-08-17:
> I did not open the PR as there is still the question of whether to have separate branches or the effort of refactoring all future PRs to be backward compatible as you suggested.

The initial PR for the rosindex is just to reserve the name of the package. Everything else can be changed later.

> I think branching `main` off into `humble` branch now might be a way to go, and then merge [#101](https://github.com/AndrejOrsula/pymoveit2/pull/101) from `dev` into `main` once all is tested.

If we cannot keep backward compatibility on the same branch, then I would also branch of old versions (`humble`) from main, and keep the latest versions (`jazzy`, `rolling`) on main. During the bloom release, you can choose which branch to release for which ROS distro.

christian-rauch, 2025-09-24:
It has been released to the ROS 2 testing repo.

You have to add the [ros-testing repo](https://docs.ros.org/en/rolling/Installation/Testing.html):
```bash
sudo apt install -y ros2-testing-apt-source
sudo apt update
```
and then you can install the package via:
```bash
sudo apt install ros-rolling-pymoveit2
```

To install the regular ROS repo again (and remove the ros-testing repo):
```bash
sudo apt install -y ros2-apt-source
sudo apt update
```

Let me know if this package works for you or if there is something missing. If this works, I will release for the other distributions too.

AndrejOrsula, 2025-09-24:
Thank you!

I did a quick test by following your instructions inside a [`ros:rolling-ros-core` Docker container](https://hub.docker.com/layers/library/ros/rolling-ros-core/images/sha256-c9129ccb9db17d0c557431feaba2dad7a6b7866648f71eaa5c17beaf1a0f64e7), but encountered this issue:

```log
root@8e10088f7573:/# /opt/ros/rolling/lib/pymoveit2/ex_pose_goal.py 
[INFO] [1758701786.878357609] [ex_pose_goal]: Moving to {position: [0.5, 0.0, 0.25], quat_xyzw: [1.0, 0.0, 0.0, 0.0]}
Traceback (most recent call last):
  File "/opt/ros/rolling/lib/pymoveit2/ex_pose_goal.py", line 147, in <module>
    main()
  File "/opt/ros/rolling/lib/pymoveit2/ex_pose_goal.py", line 102, in main
    moveit2.move_to_pose(
  File "/opt/ros/rolling/lib/python3.12/site-packages/pymoveit2/moveit2.py", line 425, in move_to_pose
    self.plan(
  File "/opt/ros/rolling/lib/python3.12/site-packages/pymoveit2/moveit2.py", line 515, in plan
    future = self.plan_async(
             ^^^^^^^^^^^^^^^^
  File "/opt/ros/rolling/lib/python3.12/site-packages/pymoveit2/moveit2.py", line 663, in plan_async
    future = self._plan_kinematic_path()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/rolling/lib/python3.12/site-packages/pymoveit2/moveit2.py", line 1931, in _plan_kinematic_path
    self._node.get_logger().warn(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'RcutilsLogger' object has no attribute 'warn'
terminate called without an active exception
Aborted (core dumped)
```

Perhaps `rolling` or even one of the previous releases introduced some breaking changes to the logging? I have not used anything beyond `humble` for a long time.

Unfortunately, I won't have time to look into this until later next month. Thanks again tho.

christian-rauch, 2025-09-25:
The package has also been released to `humble` and `jazzy` via `ros-humble-pymoveit2` and `ros-jazzy-pymoveit2`.

After:
```bash
sudo apt install ros-jazzy-pymoveit2
```
I can run:
```bash
ros2 run pymoveit2 ex_pose_goal.py
```
with:
```
[INFO] [1758825916.465877460] [ex_pose_goal]: Moving to {position: [0.5, 0.0, 0.25], quat_xyzw: [1.0, 0.0, 0.0, 0.0]}
[WARN] [1758825916.469680804] [ex_pose_goal]: Service 'plan_kinematic_path' is not yet available. Better luck next time!
[WARN] [1758825916.471608207] [ex_pose_goal]: Cannot execute motion because the provided/planned trajectory is invalid.
[WARN] [1758825916.473018347] [ex_pose_goal]: Cannot wait until motion is executed (no motion is in progress).
```