AndrejOrsula/pymoveit2 #95: wait_set error

christian-rauch, 2025-05-05:
Since https://github.com/AndrejOrsula/pymoveit2/pull/94, I am getting

```
Exception in thread Thread-1 (spin):
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 308, in spin
    self.spin_once()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 911, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 890, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 793, in wait_for_ready_callbacks
    return next(self._cb_iter)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/executors.py", line 719, in _wait_for_ready_callbacks
    if wt in waitables and wt.is_ready(wait_set):
                           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/action/client.py", line 232, in is_ready
    ready_entities = self._client_handle.is_ready(wait_set)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
rclpy._rclpy_pybind11.RCLError: Failed to get number of ready entities for action client: wait set index for status subscription is out of bounds, at ./src/rcl_action/action_client.c:642
```

when doing something simple as:

```Python
    moveit2.move_to_pose(
        position=[0.3, 0.0, 0.5],
        quat_xyzw=[1.0, 0.0, 0.0, 0.0],
    )
    moveit2.wait_until_executed()
```

This worked before and still works with the current `devel` branch.

AndrejOrsula, 2025-05-11:
Thank you @christian-rauch for opening, describing, and addressing this issue with a PR (#96)!

Hi @alonborn, when you get a moment ~ could you please check if #96 causes issues for you that you previously described in #93 and addressed in #94? Thank you.

christian-rauch, 2025-05-12:
I recently encountered issues with `wait_until_executed` blocking forever, in `rclpy.spin_once(self._node, timeout_sec=1.0)`. I assume #94 was supposed to tackle that issue?

Eventually, I worked around the issue by manually using the private `self.__send_goal_future_move_action` in `spin_until_future_complete` of my executor. But this is not ideal.

I now think that, in order to prevent such concurrency issues, all methods relying on the executors should either return a) the goal future instead and leave the execution cycle to the caller of the methods, or b) take an executor argument, which is then used by the methods to `spin_until_future_complete()`, instead of using `rclpy.spin_once()` in a loop. The executor could also be initialised in the `MoveIt2` constructor and used when a `None` executor is provided to any of the methods.

alonborn, 2025-05-14:
Hi!

Initially, I thought that PR #94 would resolve the `IndexError: wait set index too big` issue, but it turns out that it didn’t — which explains why @christian-rauch encountered it again.

This concurrency issue with the executor seems to have surfaced more recently. A few months ago, when I was using pymoveit2, everything ran smoothly without these errors.

I’ve tested the changes introduced in PR #96 — they definitely improve the situation, but I still occasionally see the exception.

As @christian-rauch also pointed out, in my opinion, the core issue is spinning the node from multiple places, which leads to inconsistent behavior. In particular, when using pymoveit2 within a service or action callback context, calling spin (or spin_once) from within a callback that was itself triggered by a spin can result in nested spin issues.

Ideally, there should be only one place responsible for spinning the node. There should be no spin_once() calls inside pymoveit2, like those that were recently introduced:
https://github.com/AndrejOrsula/pymoveit2/commit/1f030f6072dac16914c91d1368926f182ea5f1fc
By the way, I also tried using a previous version (before this change) and still encountered the exception — so there may be other underlying changes contributing to the issue.

Not sure it helps, but anyway, these are my obeservations...

AndrejOrsula, 2025-06-08:
Thank you, @alonborn, for your insights! 

> The changes are still a work in progress and not fully cleaned up yet. Once they’re in better shape, I’ll share them with you. Even if it’s not production-ready or mergeable as-is, it might still be worth reviewing. Let me know if you'd like me to polish it further for potential integration.

I think that would be very highly valuable if you have the time to look into it. Thank you!