AndrejOrsula/pymoveit2 #103: How to allow approximate IK solutions

Flamethr0wer, 2025-07-02:
Hello. I'm trying to get my MoveIt2 setup to work, and, while testing on Rviz with the MotionPlanning panel (I think due to my arms being 4-DOF), the IK planning always fails unless I set "Approx IK Solutions" to true from the panel. However, now that I'm trying to use pymoveit2 (as I'm on Humble and the official Python bindings for MoveIt2 aren't available) to control my arms, the planning is failing again:
```
[move_group-2] [INFO] [1751454070.863885775] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-2] [INFO] [1751454070.864628806] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-2] [INFO] [1751454070.887324529] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-2] [WARN] [1751454070.887539583] [moveit_move_group_capabilities_base.move_group_capability]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as start state in the motion planning request
[move_group-2] [WARN] [1751454070.887599842] [moveit_move_group_capabilities_base.move_group_capability]: Execution of motions should always start at the robot's current state. Ignoring the state supplied as difference in the planning scene diff
[move_group-2] [INFO] [1751454070.887659249] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-2] [INFO] [1751454070.903714761] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'left_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-2] [ERROR] [1751454071.404930277] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - left_arm/left_arm: Unable to sample any valid states for goal tree
[move_group-2] [ERROR] [1751454071.406774374] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - left_arm/left_arm: Unable to sample any valid states for goal tree
[move_group-2] [ERROR] [1751454071.408874970] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - left_arm/left_arm: Unable to sample any valid states for goal tree
[move_group-2] [ERROR] [1751454071.412615091] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - left_arm/left_arm: Unable to sample any valid states for goal tree
[move_group-2] [WARN] [1751454071.413029755] [ompl]: ./src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 0.508064 seconds
[move_group-2] [INFO] [1751454071.462136617] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-2] [INFO] [1751454071.463121017] [moveit_move_group_default_capabilities.move_action_capability]: Catastrophic failure
```

How can I use approximate IK solutions from pymoveit2? And if it's not possible, could this be considered as a feature?

AndrejOrsula, 2025-07-02:
Hello @Flamethr0wer,

I wonder if it is something that you can set in the config of your Kinematics solver. Some plugins allow solving only for the position. Otherwise, you could also try maxing the `orientation_tolerance` to see if that changes anything.

If not, would you go over the source code of the MoveIt RViz plugin to see what the "Approx IK Solutions" field accomplishes under the hood? I have no idea, as I never needed it.

Flamethr0wer, 2025-07-02:
@AndrejOrsula  I did manage to get it working with another kinematics plugin, but with an unusable 5cm of positional tolerance and 0.2 rads of angular tolerance, so not ideal... Do you know where the source code for the MotionPlanning panel might be?

Flamethr0wer, 2025-07-02:
I think I found something there; it appears it sets robot_interaction::KinematicOptions::RETURN_APPROXIMATE_SOLUTION, although I didn't find anything that in robot/interaction/kinematic_options_map...

Flamethr0wer, 2025-07-03:
@AndrejOrsula Do you think I could modify some parameter assignment in MoveIt2 if I built from source to always use approximate solutions? Although it'd still be nice to be able to set it from API...

AndrejOrsula, 2025-07-03:
That is certainly a possibility. I do not know if it is possible to set that opt externally via the ROS middleware, but that could also be added if you fork it.

Flamethr0wer, 2025-07-03:
@AndrejOrsula would the file to modify be move_group_interface.cpp?

Flamethr0wer, 2025-07-03:
I managed to get it! I set the default value for return_approximate_solution in kinematics_base.h to true and now MoveIt2 uses approximate solutions for any request. Although it was nice when you could easily set that programmatically with the moveit_commander for ROS1, so if you found a way to do it with pymoveit2 that'd be great. Thanks for the help.

AndrejOrsula, 2025-07-03:
Glad to hear!

AndrejOrsula, 2025-08-16:
If anyone figures out a way to enable this functionality outside of the MoveIt2 codebase, please let us know by commenting under this issue so that we can reopen it and add the functionality into `pymoveit2`. Closing for now. Thank you.