AndrejOrsula/pymoveit2 #110: README Typo & Joint Not Found Errors on ROS 2 Humble

wuxiaoqiang12, 2025-09-01:
Hi Andrej Orsula,

First and foremost, I want to express my sincere gratitude for developing and maintaining the `pymoveit2` package—it’s been an incredibly valuable tool for working with MoveIt 2 on ROS 2 humble, and I really appreciate the effort the team puts into it!

While using the package recently, I encountered two small issues that I wanted to share, in case they help improve the project for others:

## 1. README Typo: Outdated Launch File Name
In the project’s README, there’s a minor outdated reference: the launch file `ex_ign_control.launch.py` is mentioned, but it appears this file has been renamed to `ex_gz_control.launch.py` in the current codebase. This might cause confusion for new users following the documentation, so updating the README to reflect the new filename would be helpful.

## 2. Persistent "Joint Not Found" Errors When Launching on ROS 2 Humble
When running the `ex_gz_control.launch.py` launch file for the Panda robot on ROS 2 Humble, I’m getting continuous errors about joints not being found in the "panda" model. Below are the details of my setup, steps taken, and the errors encountered:

### Environment & Setup
- **ROS 2 Distro**: Humble Hawksbill
- **Packages Used**:
  - `pymoveit2`: Main branch (cloned from `pymoveit2_ws/src/pymoveit2`)
  - `panda_gz_moveit2`: `humble_devel` branch (cloned from `pymoveit2_ws/src/panda_gz_moveit2`)
- **Environment Preparation**:
  1. Deactivated all Conda environments with `conda deactivate`
  2. Activated a Python virtual environment (`.venv`)
  3. Sourced ROS 2 Humble: `source /opt/ros/humble/setup.zsh`
  4. Built the workspace and sourced the local setup: `colcon build && source install/local_setup.zsh`

### Command Run
```bash
ros2 launch panda_moveit_config ex_gz_control.launch.py
```
### Key Errors Observed
The servo_node_main and move_group nodes repeatedly log errors like this (truncated for brevity—errors repeat continuously):

```shell
[servo_node_main-4] [ERROR] [1756691897.649014352] [moveit_robot_model.robot_model]: Joint '2' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691897.649046735] [moveit_robot_model.robot_model]: Joint '4' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691897.649052526] [moveit_robot_model.robot_model]: Joint '1' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691897.649056414] [moveit_robot_model.robot_model]: Joint '3' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691897.649059970] [moveit_robot_model.robot_model]: Joint '5' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691897.649063453] [moveit_robot_model.robot_model]: Joint '6' not found in model 'panda'
```

Additionally, there are warnings about missing collision geometry for Panda links (e.g., panda_link0, panda_link1) and controller spawn failures:

```
[spawner-6] [FATAL] [1756691898.239775064] [spawner_joint_trajectory_controller]: Failed loading controller joint_trajectory_controller
[spawner-8] [ERROR] [1756691898.249958229] [spawner_joint_state_broadcaster]: Failed to configure controller
```

### Full Log
For reference, here’s the complete terminal output from the launch command:
```shell
(.venv) ➜  pymoveit2 git:(main) ros2 launch panda_moveit_config ex_gz_control.launch.py
[INFO] [launch]: All log files can be found below /home/xqw/.ros/log/2025-09-01-09-58-15-974973-h00636709-ThinkCentre-M960t-1401724
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [1401789]
[INFO] [robot_state_publisher-2]: process started with pid [1401791]
[INFO] [move_group-3]: process started with pid [1401794]
[INFO] [servo_node_main-4]: process started with pid [1401796]
[INFO] [rviz2-5]: process started with pid [1401798]
[INFO] [spawner-6]: process started with pid [1401800]
[INFO] [spawner-7]: process started with pid [1401802]
[INFO] [spawner-8]: process started with pid [1401804]
[INFO] [create-9]: process started with pid [1401806]
[INFO] [parameter_bridge-10]: process started with pid [1401808]
[servo_node_main-4] [WARN] [1756691897.191100008] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-4] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-4] to the Servo composable node in the launch file
[servo_node_main-4] [WARN] [1756691897.215851362] [moveit_robot_model.robot_model]: Link panda_link0 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[servo_node_main-4] [WARN] [1756691897.215884364] [moveit_robot_model.robot_model]: Link panda_link1 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[servo_node_main-4] [WARN] [1756691897.215888757] [moveit_robot_model.robot_model]: Link panda_link2 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[servo_node_main-4] [WARN] [1756691897.215893104] [moveit_robot_model.robot_model]: Link panda_link3 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[servo_node_main-4] [WARN] [1756691897.215898357] [moveit_robot_model.robot_model]: Link panda_link4 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[... 273 lines cut ...]
[servo_node_main-4] [ERROR] [1756691899.090347681] [moveit_robot_model.robot_model]: Joint '4' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.090350248] [moveit_robot_model.robot_model]: Joint '1' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.090351915] [moveit_robot_model.robot_model]: Joint '3' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.090353516] [moveit_robot_model.robot_model]: Joint '5' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.090355134] [moveit_robot_model.robot_model]: Joint '6' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.090320148] [moveit_robot_model.robot_model]: Joint '2' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.090347833] [moveit_robot_model.robot_model]: Joint '4' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.090352275] [moveit_robot_model.robot_model]: Joint '1' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.090355182] [moveit_robot_model.robot_model]: Joint '3' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.090357862] [moveit_robot_model.robot_model]: Joint '5' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.090360500] [moveit_robot_model.robot_model]: Joint '6' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.195845856] [moveit_robot_model.robot_model]: Joint '2' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.195867982] [moveit_robot_model.robot_model]: Joint '4' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.195870207] [moveit_robot_model.robot_model]: Joint '1' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.195871821] [moveit_robot_model.robot_model]: Joint '3' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.195873385] [moveit_robot_model.robot_model]: Joint '5' not found in model 'panda'
[servo_node_main-4] [ERROR] [1756691899.195874893] [moveit_robot_model.robot_model]: Joint '6' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.195871891] [moveit_robot_model.robot_model]: Joint '2' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.195893387] [moveit_robot_model.robot_model]: Joint '4' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.195898070] [moveit_robot_model.robot_model]: Joint '1' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.195901471] [moveit_robot_model.robot_model]: Joint '3' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.195904365] [moveit_robot_model.robot_model]: Joint '5' not found in model 'panda'
[move_group-3] [ERROR] [1756691899.195907311] [moveit_robot_model.robot_model]: Joint '6' not found in model 'panda'

```

AndrejOrsula, 2026-02-11:
I am unfortunately unsure about the other issue. It would seem that the naming of the joints is incorrect.

AndrejOrsula, 2026-09-16:
The latest release [**5.0.0**](https://github.com/AndrejOrsula/pymoveit2/tree/5.0.0) resolves this problem. Feel free to reopen this issue if necessary. Thank you for your patience.

closed 2026-09-16 by AndrejOrsula